jepegit/cellpy #8: Move other old parts of cellpy into the repo - data readers

jepegit, 2016-09-06:
The routines for reading experimental data files etc should be gradually moved into the repo. 
- data readers

torvara, 2016-09-13:
To move "data readers" into repo, start a new feature with git flow and copy them into a new folder under "cellpy" (above utils). I personally think this could be a nicer folder structure. When "data readers" are in place and working tools in "utils" can read from "data readers", finish the feature with git flow.

jepegit, 2016-09-26:
Will do. But first I will create a structure using cookiecutter with the pypackage template.

jepegit, 2016-09-26:
I have made a new structure usig cookicutter and moved my old files into it (new branch). I am not sure if I made one folder-level too much.  @torvara Could you check?

torvara, 2016-09-27:
There was indeed one level too many, but it looks good now.

closed 2016-09-27 by torvara