jepegit/cellpy #127: Get Temperature Data from .res Arbin

JohnBolander, 2019-05-09:
How can you parse the temperature data from the .res Arbin file? It does not seem that that data is contained in the output

jepegit, 2019-05-13:
Unfortunately, we have not tried to connect any temperature measurements to any Arbin instruments in our labs. I guess it is possible and it could be that the data then are stored in the same Access db table as the raw data. Maybe you can check your files if it exists any columns that contains temperatures?

A way to look directly at the loaded data is to have a look at the columns in the generated `Pandas.DataFrame`:

```python
from cellpy import cellreader
fname = fPath("Path/to/your/file/name.res")
cell = cellreader.CellpyData()
cell.from_raw(fname, mass=1.0)
data = cell.dataset.dfdata
print(data.columns)
```

This is the output from the typical files that we have:

```bash
Index(['Test_ID', 'Data_Point', 'Test_Time', 'Step_Time', 'DateTime',
       'Step_Index', 'Cycle_Index', 'Is_FC_Data', 'Current', 'Voltage',
       'Charge_Capacity', 'Discharge_Capacity', 'Charge_Energy',
       'Discharge_Energy', 'dV/dt', 'Internal_Resistance', 'AC_Impedance',
       'ACI_Phase_Angle'],
      dtype='object')
```

We could try to implement a way to parse temperature data, though we will need a file that has these kind of data in it first.

jepegit, 2020-12-22:
It is now possible to get auxiliary data. And I guess that includes T. So I will close the issue. Make a new issue if that didn't fix it.

closed 2020-12-22 by jepegit