jepegit/cellpy #150: Res file reader fails with no data for Test_ID

JohnBolander, 2019-12-14:
When trying to read a .res Arbin file, if the `Test_ID` exists in the `Global_Table` but no data exists in the `Channel_Normal_Table`, the program will fail. Why there is no data for the first test is unclear, but this case exists. Interestingly enough, the failure manifests itself in two different places depending on whether you run on Linux or Windows. On Windows `normal_dr` is set to None by default, where as on Linux you just get a dataframe with columns but no data.

jepegit, 2019-12-15:
Thanks for the feedback, John.
It is obviously best if the program behaves the same on Windows and Linux. So I think we can classify this as a bug. Not sure if it is best to let it fail, return a specified error/exception (for example define a MissingDataException), return None or return a DataFrame with the columns but with no data (that can be checked with DataFrame.empty (should return True if the DataFrame does not have any data)). I lean towards returning a DataFrame. But I am open to other suggestions.

JohnBolander, 2020-01-04:
I can understand why some users may want an error/fail, but I'd prefer to stay true to the file, there's just no data for that `Test_ID` so the DataFrame is empty. As long as `cellpy` is doing its job, then it's up to the user to track down why the file had no data. In the case I saw I suspect a test was started and stopped so quickly (for whatever reason) that no data was gathered. I don't often deal with multiple `Test_ID`s in files though and usually just extract an entire file's worth of data into memory for custom analysis or copy it over to a different database for standard viewing/analysis.

jepegit, 2021-08-28:
fixed (sorry for being very slow on fixing this)

closed 2021-08-28 by jepegit