jepegit/cellpy #166: File-specific input parameters for easyplot which ties to the batch journal

amundmr, 2021-08-12:
Input parameters on the file(/experiment/cell) level should create a new journal from batch_journals.py with data like:

- Cell state(broken/alive/used)
- Bad cycles
- Nickname
- active_mass
- nominal capacity
- Anode/Cathode material
- +++

This journal should be saveable (maybe use existing to_file function) in a format easy for the user (e.g. excel), and also re-loadable both into cellpy, and from easyplot. 

Example:
ezplt = easyplot.plot("path/to/journal.xlsx", kwargs)

jepegit, 2021-08-19:
I can have a go at this if you want @amundmr.

amundmr, 2021-08-19:
That would be great, @jepegit . I can start to colab on this later today or tomorrow. Maybe we can get a good implementation done before I leave if we work together.

jepegit, 2021-08-20:
Might be difficult to find time today - but I will give it a try.  @amundmr, maybe you should merge "your" branch into master first and then I can create a new branch? What do you prefer?

amundmr, 2021-08-20:
I have some unfinished code related to issue #167 which should not be merged with master yet, in my opinion. Maybe we can branch out from master and just handle the merge conflicts afterwords? There should really not be too much conflict. I am free throughout the day so you can just book me whenever it fits you. I see your calendar is quite busy.

amundmr, 2021-08-20:
Update: I just cleaned up most of the bad stuff, ready to merge with master, se pull request #171

jepegit, 2021-08-20:
Made a notebook (EasyPlot_Dev) in dev_utils with test data and xlsx.
Did a quick inspection and came up with the following sub-tasks that I will start to work on: 

#### Priority
- easyplot crashes if nicknames are not given - fix
- need to implement a way to export journals to xlsx
- figure out or modify easyplot so that mass and other parameters can be given to it
- figure out a way to save an easyplot session with session and metadata

#### Optional
- maybe we also need to allow for reading meta and session from xls
- should rename index to cell_name instead of filename in journal - maybe later

jepegit, 2021-08-28:
usable now - improve gradually later

closed 2021-08-28 by jepegit