jepegit/cellpy #185: Multi-channel support

streamengineer, 2021-09-12:
First of all,

thanks a lot for making this project. It is fantastic that projects like these exist. 

First of all,

thanks a lot for making this project. It is fantastic that projects like these exist. 

My problem is that with the latest version of cellpy I cannot convert .res files from a multi-channel experiment into an xls file. I usually run 16 channels in parallel with our Arbin potentiostat which show up in the Test_ID column of the Channel_Normal_Table". 

When running the following skript cellpy only outputs the first channel (test_ID = 1) to the excel file. 

`
import os
import sys

import matplotlib.pyplot as plt

from cellpy import cellreader

rawdir = (r"C:\Users\mkolen\Google Drive\PhD\Python\res files")
outFolder = (r"C:\Users\mkolen\Google Drive\PhD\Python\results")

files = ["MK-Ag-prelectrolysis-500uA.res"]

mass = 0.3

rawfiles = [os.path.join(rawdir, f) for f in files]
print("\n", "your files".center(80, "-"))

for f in rawfiles:
    exists = "OK"
    if not os.path.isfile(f):
        exists = "NOT FOUND"
    print(f"{f} {exists}")

print(80 * "-")

d = cellreader.CellpyData().from_raw(rawfiles)
d.set_mass(mass)
d.make_step_table()
d.make_summary()
d.to_csv(outFolder,sep=",")
`

Can a feature be added that enables copying of all channels of the normal table from the .res file into an excel file? 

I tried finding a parameter in cellpy that enables multi-channel usage without success. 

I'd be willing to dedicate some time to implement this feature myself but since I am a python novice I could use some guidance on how to do this. 

**Additional context**
Add any other context or screenshots about the feature request here.

In a typical experiment I run 16 channels in parallel on our Arbin potentiostat which creates a .res file that (when opened in MS Access) contains the data from all 16 channels. The channels are distinguished by the Test_ID column (Test_ID =1-16). My problem is that with cellpy I can only export the data from the first channel into an xls file. The data from the remaining 15 channels is lost.   

When running the following skript cellpy only outputs the first channel (test_ID = 1) to the excel file. 

`
import os
import sys

import matplotlib.pyplot as plt

from cellpy import cellreader

rawdir = (r"C:\Users\mkolen\Google Drive\PhD\Python\res files")
outFolder = (r"C:\Users\mkolen\Google Drive\PhD\Python\results")

files = ["MK-Ag-prelectrolysis-500uA.res"]

mass = 0.3

rawfiles = [os.path.join(rawdir, f) for f in files]
print("\n", "your files".center(80, "-"))

for f in rawfiles:
    exists = "OK"
    if not os.path.isfile(f):
        exists = "NOT FOUND"
    print(f"{f} {exists}")

print(80 * "-")

d = cellreader.CellpyData().from_raw(rawfiles)
d.set_mass(mass)
d.make_step_table()
d.make_summary()
d.to_csv(outFolder,sep=",")
`

Can a feature be added that enables conversion of .res files with data from multiple channels into an excel file? 

I tried finding a parameter in cellpy that enables multi-channel usage without success. 

I'd be willing to dedicate some time to implement this feature myself but since I am a python novice I could use some guidance on how to do this.

jepegit, 2021-09-13:
Thanks for the feedback, @streamengineer. And for the nice words.
Yes, I think we could add a feature like that. But not sure if we should open up for having many datasets within one `CellpyData`-object, though. It used to be allowed, but it soon became very difficult to maintain. 

Maybe the simplest is to have a method in the `ArbinLoader` that can return the number of test IDs it finds. And then iterate the loading/conversion (`ArbinLoader`  is the object that does the actual loading of arbin .res files).

For example:

```python

>>> info = cellpy.some.method.that.checks.rawfiles(rawfiles, instrument="arbin_res", check_test_ids=True)
>>> all_test_ids = info.test_ids
>>> for test_id in all_test_ids :
>>>    d.cellreader.CellpyData().load_raw(rawfiles, test_id=test_id)
>>>    d.set_mass(mass)
>>>    d.make_step_table()
>>>    d.make_summary()
>>>    d.to_csv(outFolder,sep=",")

```

For this to work, the following things should be implemented.

1. allow `test_id` as input to `ArbinLoader`
2. make the method (or function) that checks the arbin rawfile for the `test_ids`.

If you would like to give it a try, I will be happy to help you along the way.

jepegit, 2022-04-18:
Was not able to find the changes in the branch - only deletions. Closing issue for now.

closed 2022-04-18 by jepegit