jepegit/cellpy #191: Add support for Arbin version 7 software

jtgibson91, 2022-01-02:
Currently, the cellpy support for SQL databases is geared solely towards Arbin software version 8. That causes a problem for me (and others I'm sure) who are using the older version 7. Some differences I"ve encountered already in cellpy/readers/instruments/arbin_sql.py are database names (ArbinPro8MasterInfo for version 8) --> (ArbinMasterData for version 7). I'm trying to modify it to work for my version but am not there yet. So would appreciate any help.

jepegit, 2022-01-06:
Hi James,
That sounds like a good idea. Let me know how I best can help you.

All the best,
Jan Petter

jepegit, 2022-04-18:
I will need to have access to Arbin v7 software data to be able to work with this. I dont have that in the lab, unfortunately. @jtgibson91; if you can somehow share some data (copy of db?) I can have a look at it. But not much I can do without any data...

jtgibson91, 2022-04-27:
> I will need to have access to Arbin v7 software data to be able to work with this. I dont have that in the lab, unfortunately. @jtgibson91; if you can somehow share some data (copy of db?) I can have a look at it. But not much I can do without any data...

Hey @jepegit, I'm gonna work on getting a copy of it with some sample data so you can look at it. I have made some headway in getting the data out, I can send you the code I have so far as well so you can have a head start.

jepegit, 2023-04-07:
@chonmj - would be fantastic if you could share your code and dataframe. You can also fork and create a pull request if you want.

chonmj, 2023-04-10:
@jepegit just submitted pull request. To make your lives easier, I simply created a new instrument, "arbin_sql_7" based off the "arbin_sql" code. It still needs proper post-processing, but you should be able to query raw data from the SQL database at this point. Cheers!

jepegit, 2023-05-01:
Closing this now - think the solution by @chonmj looks good - reopen if it does not work as expected - or if anyone finds a good way how to include a test for the `arbin_sql_7 ` instrument.

closed 2023-05-01 by jepegit