jepegit/cellpy #216: Get All Channels for a Test

jayceslesar, 2022-07-22:
**Is your feature request related to a problem? Please describe.**
Ive been doing some digging with my orgs Arbin cycler and found it almost impossible to figure out what channels a test was run on...this data definitely exists in the database but doesnt seem to be useable...would like to work towards this feature when I have the code written out.

A common workflow ive found useful is for someone to pass in the name of a test and get back a file structure where each folder in that folder is a channel id and within that channel id subdir is the data for that test and channel.

Thoughts here?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jayceslesar, 2022-07-22:
I think can probably close this but would be nice to robustify the arbin sql file...getting some common access patterns would be super helpful so we can generalize exporting data from there... `TestIVChList_Table` seems to have all the channels specified for a test

jepegit, 2022-07-29:
@Ozzstein might have some thoughts / input.

jayceslesar, 2022-07-29:
I think there is a better way to figure this out....ideally would like to stratify by channel as well as not all channels in a test finish at the same time...Ive got a little suite of tools/functions ive put together for my org but dont really see a reason not to open source them once they are a little cleaned up.  The main problems are:

1. Standardizing Metadata
2. Get a good barebones export (my org doesnt trust the derivatives the cyclers calculate so no need to export those) 
3. The above (2) is blocked by somehow figuring out if a channel within a test has completed...the `DataBaseTimeIndex_Table` in database `ArbinPro8MasterInfo` doesn't seem to be 100% reliable at the moment so will need to do some digging -- current theory is that a channel within a test is complete if the `BeginOrEnd` column is. 2 for some test id and some channel id.

jayceslesar, 2022-09-20:
So I have found 0 reliable ways to determine if a channel within test has completed....still looking but if you have any ideas shoot

jepegit, 2022-09-26:
Sorry, @jayceslesar, I have no good ideas. I have not started looking into the database yet (it was @Ozzstein who worked with that feature). I hope to have some time for that soon, though.

jayceslesar, 2022-09-27:
I think the best strategy so far is to probe your database every hour or so, if the latest data in a channel is greater or equal than 3 days old we consider the channel done and dont attempt to query for it again.  This relies on the user having an external database of metadata for each test -> channel within that test, but works from a "production" perspective