jepegit/cellpy #253: reader for direct hdf5 output of individual tests from Arbin mssql database

asbjorul, 2023-03-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jepegit, 2023-03-30:
Hi @asbjorul 
Could you provide me with the specifications of the format?
Maybe you can implement it in cellpy with some help from me?

All the best,
Jan Petter

asbjorul, 2023-04-06:
A loader has been made and pushed to issue related branch. The format contains three tables that are the direct output from the Arbin MSSQL database on a MITS Pro 8 system:

"data_df" containing raw data from MSSQL table "dbo.IV_Basic_Table" with IR data from "dbo.IV_Extended_Table" from the "ArbinPro8Data_X", with the following headers:
Test_ID	Data_Point	Date_Time	Test_Time	Step_Time	Cycle_Index	Step_Index	Current	Voltage	Power	Charge_Capacity	Discharge_Capacity	Charge_Energy	Discharge_Energy	Internal_Resistance	Aux_Voltage_1

"stat_df" containing statistics data per cycle/step-combination from MSSQL table "dbo.StatisticData_Table" from the "ArbinPro8Data_X" database, with the following headers:
Test_ID	Channel_ID	Date_Time	Data_Point	Test_Time	Step_Time	Cycle_ID	Step_ID	TC_Counter1	TC_Counter2	TC_Counter3	TC_Counter4	Voltage	Current	Charge_Capacity	Discharge_Capacity	Charge_Energy	Discharge_Energy	Charge_Time	Discharge_Time	V_Max_On_Cycle	Extend1	Extend2	Extend3	Extend4	SubDataBaseName	Test_Name

"info_df" containing test info, like mass and capacity from MSSQL table "dbo.TestIVChList_Table" from the "ArbinPro8Info_X" database, with the following headers:
Test_ID	IV_Ch_ID	First_Start_DateTime	Resumed_Times	Last_Resume_DateTime	Last_End_DateTime	Schedule_File_Name	Databases	Item_ID	Grade_ID	Has_Aux	Has_Special	Schedule_Version	Log_Aux_Data_Flag	Log_Special_Data_Flag	Aux_Map	CanConfig_FileName	m_nCanConfigMD5	SMBConfig_FileName	SMBConfigMD5	Value	Value2	DataStructVersion	DataOK	SpecificMASS	SpecificCapacity	Capacity

jepegit, 2023-04-07:
Merged with master.
Nice.

closed 2023-04-07 by jepegit