jepegit/cellpy #272: [WIP] Paper Review

TomTranter, 2024-01-20:
A discussion thread for the review of the Joss paper

TomTranter, 2024-01-20:
Can you clarify why the order of the authors was chosen to be the way it is please.

TomTranter, 2024-01-20:
I have installed with conda and following the instructions here https://cellpy.readthedocs.io/en/latest/examples_and_tutorials/basics.html#the-getting-started-with-cellpy-tutorial-opinionated-version
I have copied tried loading the example data files you provide in the testdata folder but getting this error
`NoSuchModuleError: Can't load plugin: sqlalchemy.dialects:access.pyodbc`
```
import cellpy
import os
data_files = os.listdir("cellpyfiles")
for i, name in enumerate(data_files):
    print(i, name)
data_filepath = os.path.join("cellpyfiles", data_files[0])
c = cellpy.get(data_filepath)
```

TomTranter, 2024-01-20:
Seems like sqlalchemy-access wasn't installed

jepegit, 2024-01-20:
> Can you clarify why the order of the authors was chosen to be the way it is please.

Hi @TomTranter. Julia is the one that has pushed for this paper and done most of the writing. So I think she deserves to be first author. Asbjørn and Muhammad contributed less to the paper. But they have contributed more on the actual code. Then its me as the last author (the "creator" of cellpy). Last author appears to be "a thing" in several scientific fields, especially in material science / battery research. Good for me :-)

jepegit, 2024-01-20:
> Seems like sqlalchemy-access wasn't installed

Hi @TomTranter. Yes, sorry about that. `sqlalchemy-access` didn't have a `conda` package (at least last I check / or I could not find it). If I have understood `conda `correctly, all dependencies needs to be available as its own `conda `package. I thought that I should create a `conda `package for it myself, and then everything would be smooth. But have I done that? Nope.

TomTranter, 2024-01-21:
> Hi @TomTranter. Yes, sorry about that. `sqlalchemy-access` didn't have a `conda` package (at least last I check / or I could not find it). If I have understood `conda `correctly, all dependencies [...]

Ok - seems like it should be possible and I see you have the environment.yaml file with the pip section including the package but it was not referred to in the installation docs and took some time to figure out

TomTranter, 2024-02-05:
@jepegit Any further comments? - I have paused my review awaiting your response

jepegit, 2024-02-05:
Hi @TomTranter. Seems we need a bit more time than I thought. Think I have succeeded in making a `conda` package for `sqlalchemy-access`, at least. Probably need at least one more week to improve the documentation and tweak the `conda` recipe for `cellpy`. Sorry for that.

jepegit, 2024-02-27:
Summary of tasks:

- [ ] Can you clarify why the order of the authors was chosen to be the way it is please.

- [ ] I have copied tried loading the example data files you provide in the testdata folder but getting this error (NoSuchModuleError: Can't load plugin: sqlalchemy.dialects:access.pyodbc)

jepegit, 2024-02-27:
conda package for sqlalchemy-access made and order of authors clarified.

TomTranter, 2024-03-22:
Re-installed following updates and ran examples. Works for me on windows. In general I find the package a little non-intuitive to use but the classes and functions are documented well enough to figure it out. My review is finished. Thanks to the authors for making this useful package open source

closed 2024-03-22 by TomTranter