jepegit/cellpy #277: Drop support for Python 3.9 and begin supporting 3.12

ma-sadeghi, 2024-02-15:
This is a bit opinionated so I don't consider it as a show stopper, but I highly encourage you to consider it.

`celery` depends on NumPy, and they will drop support for Python 3.9 in a month ([Apr 4, 2024](https://numpy.org/neps/nep-0029-deprecation_policy.html#drop-schedule)).

jepegit, 2024-02-16:
Adding support for Python 3.12 requires some work on the `OtherPath` object since it builds on `pathlib` that had a major upgrade going from Python 3.11 to Python 3.12. Should be implemented before v1.1.

jepegit, 2024-02-23:
Closing it for now - will reopen it soon when we start making the necessary changes in `OtherPaths` for supporting py3.12

ma-sadeghi, 2024-03-15:
Let's keep this open since it's due soon, but I'll approve it for JOSS review.

closed 2024-09-20 by jepegit