jepegit/cellpy #278: Minor revisions for the JOSS manuscript

ma-sadeghi, 2024-02-15:
The paper is well-written, just needs a few minor edits. Here's the list of errors. The list was generated by ChatGPT, which I've manually reviewed and removed those that didn't seem reasonable.

- [ ] **One of the main tools in battery research are battery cycling experiments.**
   - Correction: One of the main tools in battery research **is** battery cycling experiments.

- [ ] **providing the tools to read different data formats,**
   - Suggestion for clarity: providing tools to read different data formats,
   
- [ ] **Converting those into one common data format that also includes relevant battery-specific metadata, and**
   - Correction for consistency: meta-data should be either **metadata** or **meta-data** consistently throughout the document. Here, it's mixed with "meta data" later on.

- [ ] **Several open-source libraries focus on battery test-data extraction.**
   - Consistency: test data should be either **test-data** or **test data** consistently throughout the document.

- [ ] **Provides a structured interphase for collecting and processing battery test data and exports to text format.**
   - Correction: **interface** instead of "interphase".

- [ ] **Accommodating not only data from diverse testers, but also thoughtfully embeding battery-specific metadata**
    - Correction: embedding.

- [ ] **Cellpy is implemented in python**
    - Correction: **Python** should be capitalized as it's a proper noun.

- [ ] **The cellpy-file format (usually stored in hdf5 format) contains all the data contained in the Data object together with additional relevant metadata,**
    - Consistency: **HDF5** should be in uppercase as it's an acronym for Hierarchical Data Format version 5. Also, decide on either "metadata" or "meta data" and use it consistently.

ma-sadeghi, 2024-02-18:
Hi @juliawind, thanks for addressing the issues. I just found another minor revision, but I won't clutter your issue tracker with another issue, so I'll add it here:

In the statement of need section, you've cited `galvani`, `neware_reader`, and `galv`, all of which are limited in their scope (can only process certain formats, etc.). Then, you've cited BEEP:

"BEEP (Battery Evaluation and Early Prediction [@beep]) provides a structured interface for collecting and processing battery test data and exports to text format."

but haven't provided a critiuqe of why BEEP is not good. In fact, the wording is a bit confusing because it implies that BEEP is a universal tool, as opposed to the others. It'd be ideal if you could provide a critique of BEEP so that `cellpy` is differentiated. Thanks!

ma-sadeghi, 2024-03-15:
> Hi @juliawind, thanks for addressing the issues. I just found another minor revision, but I won't clutter your issue tracker with another issue, so I'll add it here:
>
> In the statement of need [...]

This still seems to not have been addressed.

jepegit, 2024-03-22:
Wrote:

"""In addition, the Battery Evaluation and Early Prediction (BEEP) library [@beep] provides an interface
for parsing text-based data from several instruments. However, the target audience for this library seems to be
machine learning researchers, and the library appears at its current state not to be ideal for in-depth data handling
and analysis by battery researchers."""

@juliawind: Sounds OK? Or can it be interpreted as if the library is not a good library? What I tried to convey is that it seems to be not very easy to use if you just want to look and play with single data-sets, but is more useful if you have a large amount of data that you want to extract features from and train using scikit-learn.

juliawind, 2024-03-22:
@jepegit yes, sounds good! Could rephrase a bit to make it sound more positive: 
"""... However, the target audience for BEEP seems to be
machine learning researchers, focusing on handling and preparing large datasets, while cellpy provides tools for more in-depth data handling and analysis for battery researchers. """

jepegit, 2024-03-23:
Great!

closed 2024-03-23 by jepegit

ma-sadeghi, 2024-03-23:
Sounds good! Thanks for clarifying!