jepegit/cellpy #287: Example notebooks need to be updated/re-run/tested

ma-sadeghi, 2024-02-18:
- [ ] Example notebooks seem to have been updated 3-4 years ago (based on the most recent commit date). They possibly need to be updated in case the API has changed.
- [ ] Also, ideally they should be uploaded in their "saved" state with the cell outputs. This is useful in case someone just wants to study them on the web (rather than clon the repo, set up a proper environment, run jupyter, etc.)
- [ ] This one's optional, but highly recommended: `cellpy` seems to offer plenty of functionality, but you have very few example notebooks, which leaves users the option to comb thorugh the API, which is reasonable for most software, but probably not for research software (because the audience are mostly domain experts with far less computer expertise). I suggest you a add a few more examples to demonstrate what the software does and is capable of.

jepegit, 2024-02-26:
**Note to self**: Need to install `cellpy` in the action to be able to run notebooks using `nbmake`.
**Another note to self**: `pandas `2.2 does not handle `df.melt` as tolerant as `pandas `2.1 (multiindex columns) - need a fix for this for future versions (`pandas.stack`?)

jepegit, 2024-02-27:
Added a few and will add more in the future.

closed 2024-03-15 by ma-sadeghi