jepegit/cellpy #290: API is not documented

ma-sadeghi, 2024-02-18:
I might have missed it, but I couldn't find the API reference in your documentation. The CLI though was documented. If `cellpy` is mostly a CLI tool, then I think it should be okay. But, if it's also to be used inside scripts, you really need an API reference section in your documentation. Sphinx can automatically generate nicely formatted API reference from your function/class docstrings if they adhere to a some popular formats (like [NumPy style, or Google style](https://www.sphinx-doc.org/en/master/usage/extensions/napoleon.html), and maybe more)

jepegit, 2024-02-18:
The API reference is only accessible on the bottom of the first page (README) in readthedocs ("Indices and tables"). This makes it unnecessarily difficult to find. This should be improved, e.g. by making it accessible in the sidebar.

ma-sadeghi, 2024-02-18:
Oh right! My bad, I missed it. 

Correct me if I'm wrong: It seems that the API reference as it stands only documents your exceptions and warnings. I wasn't able to find function/class/module references.