jepegit/cellpy #301: Running example scripts (installation issues)

MaximevdHeijden, 2024-03-11:
Dear cellpy team, 

I still encounter problems when running an example script (01_laoding_data).
I get the following error: "NoSuchModuleError: Can't load plugin: sqlalchemy.dialects:access.pyodbc", which is related to the odbc driver installation. I checked your documentation again and you do refer to this (for Windows users) by provinding a link to the website to install the driver. However the weblink does not work. 
After installing the driver and updating sqlalchemy (installing the requirements.txt also did not work), it still does not work on my Windows machine. 

The example also does not work with Linux as I get the following error: InterfaceError: (pyodbc.InterfaceError) ('IM002', '[IM002] [Microsoft][ODBC Driver Manager] Data source name not found and no default driver specified (0) (SQLDriverConnect)')
(Background on this error at: https://sqlalche.me/e/20/rvf5)"". I did install the mbdtools as referred to in the documentation and I should have the correct version of sqlalchemy as I installed the requirements.txt file.

Can you please fix this and try to install cellpy and run it on another machine? Thanks!

Can you also still add jupyter notebook to the requirements.txt file (and refer to it in the installation)?

This issue is related to https://github.com/openjournals/joss-reviews/issues/6236

MaximevdHeijden, 2024-03-11:
I realized that my python and drives dont have the same bit on that PC so I tried the installation on another PC as well (Windows). 
There it seemed to run the script AFTER specifically installing the requirements.txt file. Can you add this to the installation please?

MaximevdHeijden, 2024-03-11:
Regarding the same example script, "tables" needs to have a specific version (at least 3.7). please add this to the requirements.txt

MaximevdHeijden, 2024-03-11:
03_capacity_vs_voltage: required the installation of the kaleido package

MaximevdHeijden, 2024-03-11:
02_Initial_data_inspection: somehow it does not recognize the .summary_plot function for me, although its in the plotutils script.

MaximevdHeijden, 2024-03-11:
04_incremental_capacity_analysis: the ica.dqdv(c) does not run, I get the following error: "TypeError: dqdv() missing 1 required positional argument: 'capacity'"

MaximevdHeijden, 2024-03-11:
Please fix the above issuesand rerun all examples scripts. Once you have addressed the issues, the software is ready to use as far as I'm concerned!

juliawind, 2024-03-13:
link to access drivers fixed and reference to requirements.txt added in installation notes.
(issue with dqdv is fixed the master branch)

jepegit, 2024-03-15:
> Dear cellpy team,
> 
> I still encounter problems when running an example script (01_loading_data). I get the following error: "NoSuchModuleError: Can't load plugin: sqlalchemy.dialects:access.pyodbc", which is related to the odbc driver installation. I checked your documentation again and you do refer to this (for Windows users) by provinding a link to the website to install the driver. However the weblink does not work.
> 

Updated the link :-)

> The example also does not work with Linux as I get the following error: InterfaceError: (pyodbc.InterfaceError) ('IM002', '[IM002] [Microsoft][ODBC Driver Manager] Data source name not found and no [...]

I forgot to make a conda package of the latest `cellpy` version. Now it is made.
Tried it on several linux distros (ubuntu and RHEL) without being able to reproduce the error. The intended behavior of `cellpy` is to check if it is on windows or posix first by itself and then only use `sqlalchemy `and `pyodbc` if it is on windows, while using `mdbtools` and `pandas.read_csv`, when not on windows.  Could be that the error was fixed "unintentionally" while fixing another error so that it works on the newest version while not on the older one...

> 
> Can you please fix this and try to install cellpy and run it on another machine? Thanks!

So, not sure if I have fixed it, but at least it runs on the ubuntu and RHEL linux distros I have access to.

> Can you also still add jupyter notebook to the requirements.txt file (and refer to it in the installation)?

Yes, added `jupyterlab` and jupyter `notebook` to requirements.txt

> This issue is related to [openjournals/joss-reviews#6236](https://github.com/openjournals/joss-reviews/issues/6236)

jepegit, 2024-03-15:
> 04_incremental_capacity_analysis: the ica.dqdv(c) does not run, I get the following error: "TypeError: dqdv() missing 1 required positional argument: 'capacity'"

This should now have been fixed in v1.0.1b5 for both pip and conda

jepegit, 2024-03-15:
> Regarding the same example script, "tables" needs to have a specific version (at least 3.7). please add this to the requirements.txt

Added to requirements.txt

jepegit, 2024-03-15:
> I realized that my python and drives dont have the same bit on that PC so I tried the installation on another PC as well (Windows).
> There it seemed to run the script after specifically installing the requirements.txt file. Can you add this to the installation please?

Added "hint" about installing requirements using the "requirements.txt" file. 
Added link to note about need of same bit on driver and python there as well.

jepegit, 2024-03-15:
> 02_Initial_data_inspection: somehow it does not recognize the .summary_plot function for me, although its in the plotutils script.

Runs well now

jepegit, 2024-03-15:
> 03_capacity_vs_voltage: required the installation of the kaleido package (pip install -U kaleido)

Now a part of the conda package as well and referenced in the "installation guide" (under "Dependencies")

jepegit, 2024-03-15:
> Please fix the above issuesand rerun all examples scripts. Once you have addressed the issues, the software is ready to use as far as I'm concerned!

All fixed and all examples scripts were rerun also on linux without crashing :-)

closed 2024-03-25 by MaximevdHeijden