jepegit/cellpy #303: Add a number of features to the cycle statistics data

agerwick, 2024-03-11:
We would like to add a number of features on cycle statistics.
Most are just simple aggregations (Min/Max/Mean), Total Charge/Discharge/Cycle/Rest time, etc., but we also have some more ambitious goals :)
We would be happy to help adding these features.

jepegit, 2024-04-25:
@agerwick, I am moving this to a later release milestone. Need to make a v1.0.2 release very soon - there has been an accumulation of small bugs in the batch-processing utility that needs fixing...

agerwick, 2024-04-25:
@jepegit No worries, I didn't get time to work on it yet, but it's coming...

valentinsulzer, 2024-05-02:
More generally, is it possible to customize the summary features locally by passing in arguments to `make_summary`, or is the only way to overwrite the `cellpy` code?

jepegit, 2024-06-28:
I guess we can conclude that the `make_summary` method needs to be updated:

- [ ] #315 and this issue: more flexibility wrt sending arguments to the method
- [ ] #313 allowing for additional procedures to run during summary creation, and in particular estimating IR from ocv relaxations.

Starting with #315 probably makes most sense? Or maybe both should be implemented within one branch?

morrowrasmus, 2024-07-31:
After having given this some more thought, I don't think #315 is the way to achieve what we want after all. In the `summary`-table, the calculations are explicitly done (mostly in the `_generate_absolute_summary_columns()`-method), and keeping additional columns would only be a pre-requisite to calculate more values for the `summary`-table and not achieve anything by itself. It might be a idea to still implement #315 so the user has a way to receive a cleaner output of choice, but from our point of view that would not be a priority (we can clean up the table later). Maybe @valentinsulzer has some input on what he would want included that might change that?

Instead of the user specifying the columns to keep, I think what we really want to do is to expand both the `steps` and `summary`-tables to include more things (and thus also expand the two lists of columns to keep for the two tables, respectively, as needed):

`steps`: 
- include at aggregations of energy and power as well (and maybe any other values?)
- include weighted averages (as in #314)

`summary`: 
- include more simple aggregations as @agerwick suggested, for example min/max potential and initial/end potential on charge/discharge (can be added directly from the `steps`-table). This does create some redundancy as the data will be present in both the `steps` and `summary` table, but it provides (in my opinion) a more convenient way of accessing this data
- include the CV-share and other CV-share-related metrics (following the implementation of #312)
- include calculated IR and related values (following implementation of #313)
- include time-based values (e.g. total charge time, discharge time, rest time etc. for the cycle)

Apart from the previous issues on CV-share and IR-calculations, I think most of this should be fairly straightforward to implement (maybe except the weighted averages which I struggled to implement into the call to `.agg()` when I played around a little with it).

Maybe we should close #315, keep this issue open for the changes to the `summary`-table and create a new issue for the changes to the `steps`-table?