jepegit/cellpy #320: Analysing Maccor binary data

ITEllis, 2024-08-22:
I'll precis this by saying I am quite inexperience with Python but I have been messing around with Python in an attempt to (semi)-standardise how we process our cell cycling data.

Using a .dll provided by Maccor it is possible to interrogate and export binary maccor data, and I have successfully used it to take Maccor data into a dataframe or a .csv. 

Advantages of using the binary file over the Maccor exported data are that the binary file has an extractable copy of the test procedure in it, and variables like C rate and mass can be extracted (rather than manually re-associated with the file). 

What I was trying to do was understand what structure the columns and header data needs to be in for cellpy just to work with it/convert it into the cellpy format and then use cellpy to do my graphing rather than me try and re-invent the wheel.

If there is a guide in the docs that I have missed in formatting data correctly let me know, I am also happy to share what I have - it should be self-explanatory what is going on. 

Thank you for your help.

jepegit, 2024-09-04:
Hi @ITEllis,
Sorry for being slow in answering (been busy days at work).  The column-names etc can be defined in an instrument-file (yaml-format) if you want to load from for example a .csv file. It is perhaps not what you were asking for, though. If you could share your code I can have a look and (if we are lucky) I can come up with a suggestion.

ITEllis, 2024-09-04:
Hi @jepegit 
Not to worry, I asked in August and assumed the team would be on leave enjoying summer anyway.
I have attached what I have excluding the .dll as I think that is proprietary to Maccor (I'd need to check). It should be quite self explanatory what is happening. In the Jupyter notebook there is section where it uses the .dll to read the Maccor data into a dataframe with column names that I pick, plus some extra information that you don't get if you export the file using the Maccor functions (e.g. it can write out the test procedure). 

Rather than export to .csv and then use a template to load it into cellpy, I was hoping to take the df and analyse with cellpy directly - provided I labelled things in the correct way. Alternatively, exporting it into the standard cellpy h5 format and then reading in would also work. 

Let me know your thoughts, and thank you for your help

[ReadBinaryMaccor.zip](https://github.com/user-attachments/files/16868480/ReadBinaryMaccor.zip)

ITEllis, 2024-09-30:
Thank you very much, I haven't had much of a chance to have a proper play yet.
I'll try and get it to read the Maccor steps into a step table as a bonus action, but most of my testing is standardised anyway.
Thanks again