jepegit/cellpy #324: CIFS File System Causes [Errno 95] Operation not supported Error in cellpy

Kraziyi, 2025-01-16:
Describe the bug
When using cellpy to read .res files stored on a CIFS-mounted network drive, the program crashes with an error related to unsupported extended attributes. This issue seems to stem from cellpy attempting to use os.listxattr, which is not supported by the CIFS file system in certain configurations.

Cellpy version: 24.3.1
Python version: 3.11.10
Operating System: Linux

To Reproduce
Steps to reproduce the behavior:
Mount a network drive using CIFS.
Attempt to read an Arbin .res file stored on the CIFS-mounted drive using cellpy.
Observe the error traceback.

Example code:
from cellpy import cellreader
file_path = "/path/to/cifs/mounted/network/drive/file.res"
cell = cellreader.get(file_path, instrument="arbin_res")

Error traceback:
Traceback (most recent call last):
  ...
  File "/usr/lib/python3.11/shutil.py", line 384, in copystat
    _copyxattr(src, dst, follow_symlinks=follow)
  File "/usr/lib/python3.11/shutil.py", line 326, in _copyxattr
    names = os.listxattr(src, follow_symlinks=follow_symlinks)
OSError: [Errno 95] Operation not supported: '/path/to/cifs/mounted/network/drive/file.res'

Expected behavior
The .res file should be processed without errors, even if the file system does not support extended attributes. Ideally, cellpy should provide an option to bypass extended attribute operations or handle this limitation gracefully.

Desktop (please complete the following information):
OS: Linux (e.g., Ubuntu 22.04)
Version: Kernel 5.x
Additional context
This issue occurs due to the lack of support for extended attributes (xattr) in CIFS file systems by default.
Temporary Workarounds:

Copy the .res file to a local file system (e.g., Ext4, NTFS) before processing.
Modify the copy_to_temporary method in cellpy to bypass extended attribute checks:
self._temp_file_path = self.name

Proposed Solutions:
Add a configuration option to disable or bypass extended attribute operations in cellpy.
Improve error handling to detect unsupported file system operations and suggest workarounds.

jepegit, 2025-02-19:
@Kraziyi, does this work?

```python

cell = cellreader.get(file_path, instrument="arbin_res", refuse_copying=True)
```

jepegit, 2025-04-03:
Closing this now due to inactivity.

closed 2025-04-03 by jepegit