jepegit/cellpy #332: AttributeError: module 'pathlib' has no attribute '_windows_flavour' on import with Python 3.12

jbrizuela17, 2025-04-15:
**Describe the bug**
Mamba reports incompatibility with cellpy version 1.0.2 when attempting to install it in a Python 3.12 environment. This contradicts the release notes for cellpy 1.0.2, which explicitly state added support for Python 3.12.

*cellpy version: 1.0.2
*Python version:  3.12.10
*Operating System: Windows

**To Reproduce**

```
input:
mamba install cellpy=1.0.2

output:
error     libmamba Could not solve for environment specs
    The following packages are incompatible
    ├─ cellpy =1.0.2 * is installable with the potential options
    │   ├─ cellpy 1.0.2 would require
    │   │   └─ python >=3.10,<3.11.0a0 *, which can be installed;
    │   └─ cellpy 1.0.2 would require
    │       └─ python >=3.11,<3.12.0a0 *, which can be installed;
    └─ pin on python 3.12.* =* * is not installable because it requires
            └─ python =3.12 *, which conflicts with any installable versions previously reported.
critical libmamba Could not solve for environment specs
```

**Expected behavior**
cellpy version 1.0.2 should install successfully in a Python 3.12 environment, as indicated by the release notes which state: "Dropped support for python 3.9 and added support for python 3.12 (and probably beyond) by upgrading OtherPaths".

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

jepegit, 2025-05-05:
Working on fixing the conda package (it was currently restricted to python 3.10 and python 3.11)
Use either pip or uv to install until it is fully resolved (issue #333)

jepegit, 2025-05-14:
Hopefully fixed now (issue #333).

closed 2025-05-14 by jepegit